shichaoy/cube_slam #3: kitti odometry datase

iunknown10, 2019-05-29:
Hi shichaoy
For kitti odometry dataset , How to set the  transToWolrd matrix?
Thanks.

shichaoy, 2019-05-31:
@iunknown10  transToWorld is the transform matrix from camera to world ground frame. In the first frame, we can assume axis are aligned, therefore, transToWorld is [1 0 0 0;0 0 1 0;0 0 -1 1.7;0 0 0 1], namely camera is 1.7m above the ground, parallel to ground. But when camera moves, it changes.

drscopus, 2019-11-14:
@shichaoy Do you mean that the camera axis are aligned to the axis of world ground frame in the first frame?  Are the camera axis parallel to the axis of world ground frame? Is camera_x aligned to world_z, camera_y aligned to world_y, camera_z aligned to world_x?   And in Tum cabinet(cuboid) data this is also same, right? Thanks!

shichaoy, 2019-11-19:
see second page of https://github.com/shichaoy/matlab_cuboid_detect/blob/master/illustrations.pdf
In first frame KITTI data, assuming cam-x align world-x  cam-z align world-y .  that's why we have
transToWorld  (T_world_cam) = 
[1 0 0 0;
 0 0 1 0;
 0 -1 0 1.7;
 0 0 0 1],

closed 2019-11-19 by shichaoy