shichaoy/cube_slam #4: How to run it with my own image sequences

FatEvilCat, 2019-06-26:
Hi shichaoy,
Thanks for your sharing. I have some question on how to run it on my own image sequences.
I found that in object_slam/data, there is some other inputs(not the raw_imgs). I suppose that they can obtain by other packages. As a beginner, I am confused of the usage of other packages. So can you tell me how to create the stuff in object_slam/data briefly(especially the yolo in 2d_object_detect)?
Thanks in advance.

shichaoy, 2019-06-30:
hi, pop_cam_poses is the camera poses to ground (as mentioned in readme, roll/pitch/height is useful).   In current code of toy datasets (without orbslam to provide pose), you can just use a fixed camera pose parallel to ground, the single image ```cube_detection ``` (https://github.com/shichaoy/cube_slam/blob/master/detect_3d_cuboid/include/detect_3d_cuboid/detect_3d_cuboid.h) can sample and estimate camera roll/pitch, but may not be accurate.     In the future, we will upload code integrated with orb slam, then this pose can be directly provided by SLAM estimation.

zhangjun-xyz, 2019-08-07:
@shichaoy Do you have any specific plans on publish your code integrated orb slam?

shichaoy, 2020-10-05:
@zhangjun-xyz  forget to mention, already integrated into orb slam.

closed 2020-10-05 by shichaoy