shichaoy/cube_slam #5: Weights for different costs in the optimization

moshanATucsd, 2019-06-27:
Hi, thanks for the code! It says in the paper `From our
experiments, object-camera and point-camera measurements
have similar weights`, I am wondering where are those weights? I found this in the code, is this the weight? 

```
		Vector9d inv_sigma;inv_sigma<<1,1,1,1,1,1,1,1,1;
		inv_sigma = inv_sigma*2.0*cube_landmark_meas->meas_quality;
		Matrix9d info = inv_sigma.cwiseProduct(inv_sigma).asDiagonal();
```

shichaoy, 2019-07-03:
hi thanks for the interest. yes, parameter 2 can be tuned, but is in the same magnitude from our tests.

moshanATucsd, 2019-07-03:
I am wondering what's the magnitude, 0.01, 0.1, 1 etc?

shichaoy, 2019-07-04:
From my experience, usually between 1-5.

moshanATucsd, 2019-07-04:
Got it!

closed 2019-07-04 by moshanATucsd