shichaoy/cube_slam #9: Set Rviz in ROS Kinetic

godblezzme29, 2019-09-02:
Hello I have some question related to the cube_slam
1. I have finished the installation and tried to run the example, when I run "roslaunch object_slam object_slam_example.launch" on the command, I only see this result :
read data size:  51  58  58
Finish all optimization! Begin visualization.
Finish all visulialization!
Is there any way how to visualize the result? my ROS version is kinetic

2. Is it possible to test by my own image?

Thank you

shichaoy, 2019-09-06:
hi I have a section in Readme talking about this. https://github.com/shichaoy/cube_slam#notes

closed 2020-10-05 by shichaoy

yipingp, 2023-05-11:
> hi I have a section in Readme talking about this. https://github.com/shichaoy/cube_slam#notes

Hi, how do you collect truth_cam_poses? The first frame of it seems to be the transformation to the ground coord, and important to 3d cuboid detection? How should I get pose relative to ground?