shichaoy/cube_slam #10: how to get other seven vps

dongmiller, 2019-09-04:


shichaoy, 2019-09-06:
hi  you can look at Fig2 (a) and Sec III-A2 in the paper, if three Vp are known, and corner 1 is known. Then it's not difficult to get all other corners through line intersections. In the paper I already mention 5 corners.

dongmiller, 2019-09-06:
Thanks for your reply. I can get the 8 corners in the fig2(a) and (b). However,In the fig2(c), I can get the 4 corners of the visibale face, but can’t get the 4 corners of invisible face, as there isn’t any intersection of lines can be used?~

发自我的iPhone

------------------ Original ------------------
From: Shichao Yang <notifications@github.com&gt;
Date: Fri,Sep 6,2019 1:02 PM
To: shichaoy/cube_slam <cube_slam@noreply.github.com&gt;
Cc: dongmiller <944201213@qq.com&gt;, Author <author@noreply.github.com&gt;
Subject: Re: [shichaoy/cube_slam] How to get other seven corners in one face? (#10)

shichaoy, 2019-09-11:
For fig2(c), the visible four corners are easy to compute, no need to sample. We need to sample a hidden top corner, for example corner 2 in https://github.com/shichaoy/matlab_cuboid_detect/blob/master/illustrations.pdf    then we can compute all other seven corners. You can also see code here https://github.com/shichaoy/matlab_cuboid_detect/blob/74727788a0b7222546eafcf012f89e336ebb5611/detect_cuboid.m#L555

dongmiller, 2019-09-11:
> For fig2(c), the visible four corners are easy to compute, no need to sample. We need to sample a hidden top corner, for example corner 2 in [...]

I know it now, thank you for your repley!

closed 2019-09-11 by dongmiller