shichaoy/cube_slam #15: What is the mathematical basis for computing the movement of camera?

drscopus, 2019-11-11:
Hi, thank you for your great contribution! 
I am reading the source code of object_slam. I found that in you code the movement of camera is related to the variable "odom_val". This variable is computed  with two other variables "prev_pose_Tcw" and "prev_prev_pose_Tcw". Then  the current pose of camera, i.e. "curr_cam_pose_Twc" can be got. 
But both "prev_pose_Tcw"  and "prev_prev_pose_Tcw" are computed by optimization using g2o, i.e. "SE3Vertex->estimate()".
It is well-known that image matching and PnP is a good method to compute the relative movement between two images. What is the theory basis in your code? Thanks!

moshanATucsd, 2019-11-15:
Hi, I think Eq. 7 in [paper](https://arxiv.org/pdf/1806.00557.pdf) may answer your question

drscopus, 2019-11-18:
> Hi, I think Eq. 7 in [paper](https://arxiv.org/pdf/1806.00557.pdf) may answer your question

Thank you for your reply!  I think this optimization result doesn't  have scale, i.e. the scale is still not clear, right?

moshanATucsd, 2019-11-18:
As stated in the paper, for ground objects, `The scale is determined by the camera height in the
projection process`, and `We also scale ORB SLAM’s initial map by the first frame camera height (1.7m in our implementations) in order to evaluate its absolute pose error`. 

So I think the scale is defined via prior camera height.

closed 2019-11-19 by drscopus