shichaoy/cube_slam #16: How to evaluate 3D IOU for Seq. 93

moshanATucsd, 2019-11-15:
Hi Shichao,

In your CubeSLAM paper, table below, there's 3D IOU for Seq. 93. However, I can't get the correct tracklet and trajectory alignment, as the car seems to be going downwards. 

I am wondering whether you could advice how to process the tracklet file correctly? Eg. how do you load the tracklet and plot the bounding boxes, and whether have you set the camera height to some fixed value? It would be even better if you could kindly share your code for evaluation of 3D IOU for Sq. 93 

[image]

Here's what I get. bounding boxes are on the horizontal plane,

[image]

but trajectory is downwards.

[image]

shichaoy, 2019-11-18:
Hi Thanks for the interest. Evaluation on KITTI is quite complicated as there are several datasets, including KITTI object, kitti raw sequence, odometry benchmark, tracking datasets. I am using different methods to get ground truth and further evaluation. Here are some of my explanation: Not sure if it answer all of your questions.

1. I remember tracklet file is only for dynamic cars. so most kitti raw sequences have empty tracklet file. tracklet file is for kitti tracking dynamic sequences. Used in my dynamic slam section.

2. For Table III KITTI raw sequence, there is actually a mapping from KITTI object images to KITTI raw sequences. therefore we can get some selected images in KITTI raw sequences that have object annotations.

3. For non-odometry benchmark sequences (raw sequences + tracking sequences), I use gps information as pose ground truth. There is dev kit on KITTI.

4. I remember tracklet file contains object ID and positions in each frame. There is dev kit on website to read them.

5. For all object 3D IoU evaluation, I evaluate object position relative to some camera frame, not the global world frame. This is because the ground truth object annotations are also with respect to local camera frame. After getting detected and truth cuboid all in local camera/ground frame, I compute the IoU between them.

6. I do use a prior camera height ~1.7m, but not fixed it in BA optimization. I don't have much content of your GUI, you might need to look at more carefully the axes directions. For example, normally we define x right  y downward, z forward in SLAM, but it maynot be the case of KITTI truth data.

moshanATucsd, 2019-11-18:
Hi, thanks so much for your detailed reply! For point #2, the object annotations are per frame, instead of one object id for the same objects for all frames. So we need to evaluate on a per frame basis, and average the 3D IOU, is that right? You also said in point 5 that IOU is in camera frame. My only question is that should we take average for all IOU values. 

For point 3 and 6, I think 0093's GPS is wrong, as in [issue](https://github.com/utiasSTARS/pykitti/issues/18). Frankly speaking I think we have to exclude 0093 from the experiments.

closed 2019-11-19 by moshanATucsd