shichaoy/cube_slam #18: How to run the mono_dynamic.launch successfully?

drscopus, 2019-11-22:
We want to learn how you deal with dynamic objects by reading related codes in your great work. But now we can not run mono_dynamic.launch successfully. Could you give us more instructions and  provide some example data like seq07 to run it successfully? Thanks!~

drscopus, 2019-12-03:
@shichaoy hi, could you give any instructions on how to run mono_dynamic.launch? Any instruction is appreciated.  Thanks.

drscopus, 2019-12-04:
@shichaoy Would you like to share the mask image mentioned in the code of Frame.cc for removing dynamic features.
 `       std::string pred_mask_img_name = base_data_folder + "/rcnn_labelmap_3dmatched/" + frame_index_c + "_maskmap.png";`

shichaoy, 2019-12-05:
hi for dynamic slam data, see here https://drive.google.com/open?id=1T2PmK3Xt5Bq9Z7UhV8FythvramqhOo0a

drscopus, 2019-12-19:
Thank you very much!

closed 2019-12-19 by drscopus