shichaoy/cube_slam #20: Run on my own sequence

lykius, 2019-12-06:
Hi,
Your work looks amazing and I'd like to try it for one project I'm working on.
Unluckily, I can't figure out how to run your work on one of my sequences.
I managed to compile everything with ROS and also to run the object_slam example.
Now, I have a sequence of images from a driving scene (with dynamic objects) and I'd like to obtain SLAM and 3D object detection results. I also have camera intrinsincs. 
Would you be so kind to provide instructions to run cube_slam on a sequence of images?
Thanks in advance for your support!

ns15417, 2020-09-22:
@lykius Hey， how about your project now? I met the same problem now and did you figure out how to use cubeSlam on your own data?

lykius, 2020-09-28:
Hi @ns15417,
Eventually we decided to run cube_slam out without ROS.
Have a look at our fork: https://github.com/CVLAB-Unibo/cube_slam.

shichaoy, 2020-10-05:
@lykius  sorry, I sometimes missed the issues. actually we can also convert images into a bag file, just a small cpp file. I have it locally, let me know if you are interested.

closed 2020-11-09 by shichaoy