shichaoy/cube_slam #21: process has died when run orb_object_slam

GuaguaguaLiu, 2019-12-10:
Hello, thanks for your great job and open source sharing.
I try the command ‘roslaunch orb_object_slam mono_dynamic.launch’, but there is a problem that '[ros_mono-3] process has died [pid 10101, exit code -11...'
I don't know why. Look forward to your reply. Thank you!
[image]
[image]

closed 2019-12-10 by GuaguaguaLiu

MinaHenein, 2019-12-12:
Hey @GuaguaguaLiu 
I'm facing the same error, could you fix it? and how? please

GuaguaguaLiu, 2019-12-16:
Hello，I think you may need to get the 'ORBvoc. bin' in "cube_slam/orb_object_slam/Vocabulary/"

MinaHenein, 2019-12-17:
Thanks! All good now

JackingW, 2021-04-19:
Hello, @GuaguaguaLiu 
when I tried the command ‘roslaunch orb_object_slam mono.launch’, there was a problem about 'malloc(): memory corruption' like the following pictures:
[image]
[image]

whether you faced the error before?
Is it because the  'ORBvoc. bin' ?

but when I run './Vocabulary/bin_vocabulary', this problem './Vocabulary/bin_vocabulary: error while loading shared libraries: libDBoW2.so: cannot open shared object file: No such file or directory' occurred.

Looking forward to your response to the above questions. Thank you!