shichaoy/cube_slam #23: How to get the data /pred_3d_obje_matched_tracked_txt/frame_id_3d_cuboids.txt?

GuaguaguaLiu, 2019-12-16:
Hello, shichao, thanks for your great job and open source sharing.
When testing the object_orb_slam (run mono_dynamic.launch), the data you provided (kitti (0018)) is required. I see the format of the 3D_cuboids data in  /pred_3d_obj_matched_tracked_txt/frame_id_3d_cuboids.txt, i.e., [cuboid_center, yaw, cuboid_scale, [x,y,w,h], prop, truth_id]. I know the the [x,y,w,h] can be provided by YOlO or other detectors, but how  should the 'cuboid_center, yaw, cuboid_scale, and truth_id' data be obtained? Can by the detect_3d_cuboid package?

MinaHenein, 2019-12-18:
I have the same question, any luck?
the detect_cuboid doesn't save any cuboids.txt file

shichaoy, 2019-12-19:
hi I didn't upload the script, but in the matlab cuboid detection https://github.com/shichaoy/matlab_cuboid_detect/blob/master/detect_cuboid.m
At the end, the best object is stored in ObjectSets, then its straightforward to save to txt like the following:

    frame_pred_object_info_mat = zeros(0,12);
    one_obj_info = zeros(0,12); % original 11, now 12, with confidence    
    for ii=1:size(ObjectSets,1)
        if ( size(ObjectSets{ii},1)>0 )
            pred_cuboid = ObjectSets{ii}(1); % retrieve the best cuboids from each object sets
            one_obj_info(1:3) = pred_cuboid.pos;
            one_obj_info(4) = pred_cuboid.rotY;
            one_obj_info(5:7) = pred_cuboid.scale;  % half width/length/height
            one_obj_info(8:11) = pred_cuboid.rect_detect_2d;
            one_obj_info(12) = pred_cuboid.confidence_2d_prob;
            frame_pred_object_info_mat=[frame_pred_object_info_mat;one_obj_info];
        end
    end

     dlmwrite(save_pred_cuboids_txt_name,frame_pred_object_info_mat,'delimiter','\t','precision',3);

I will update the back to have this small section.  basically not matter which 3D object detection algorithm, you can do something similar to above, save to txt, feed to slam.

GuaguaguaLiu, 2019-12-19:
Thank you for your reply!

lin-fangzhou, 2023-04-19:
In the mono. launch file, there are these two parameters.

mono.launch # save result
param name="whether_save_online_detected_cuboids" value="false" /
param name="whether_save_final_optimized_cuboids" value="false" /

What I understand is that setting them to true will save the coordinates of the cube of the object in the SLAM map. But after I set them to true, I still couldn't find any results. Can someone help me?