shichaoy/cube_slam #25: cuboid_box_size, heading_angle, center for the detected cuboid.

N-G17, 2020-01-13:
Hye shichao,
I am using detect_3d_cuboid (C++ version) for single image cuboid detection. I wish to save the  cuboid_box_size, heading_angle and center for the detected cuboid in order to compute 3D IoU for the detected object in the test image. Can you please suggest the changes to be made in the code?

shichaoy, 2020-01-28:
hi the output of detection function is class object defined in https://github.com/shichaoy/cube_slam/blob/1687e7cca6e77c96a507655c8dd6c1e7653535ae/detect_3d_cuboid/include/detect_3d_cuboid/detect_3d_cuboid.h#L15
pos/scale/rotY  defines the cuboid center position/cuboid 3D size/heading,   all in ground frame, which can fully define the cuboid's 3D pose.   You can save them to txt then evaluate.

N-G17, 2020-01-28:
Thank you for your response @shichaoy. 
Yes, I figured that out after going through the code again.
Thanks.

closed 2020-02-04 by N-G17