shichaoy/cube_slam #27: Dynamic SLAM

GuaguaguaLiu, 2020-01-19:
Hello, shichao, thanks for your great job and open source sharing.
In 'mono dynamic.launch' file, I find that  the values of "‘use_dynamic_klt_features’, 'triangulate_dynamic_pts', 'ba_dyna_pt_obj_cam' and 'mono_allframe_obj_depth_init' are all set to 'false', and thus the optimization process does not consider the dynamic point observation. According to your paper, this term is involved in the optimization.  I try to set all the above parameters to be 'true', and the result is poor (especially the trajectories of the dynamic objects). So I want to know what is possible reason? In addition, what is the meaning of the parameter 'mono_allframe_obj_depth_init'? Look forward to your reply. Thank you!

shichaoy, 2020-01-28:
hi @GuaguaguaLiu   I think the main reason is the weight parameter of difference costs, for example change object_velocity_BA_weight=1. Frankly speaking, in current implementation, having these feature turning on doesn't have much effect in final metrics from my limited tests, but will make the parameters more sensitive. That's why I turn them off by default.

mono_allframe_obj_depth_init means whether directly initialize point positions through single image 3D cuboid detection, in EVERY frame.

GuaguaguaLiu, 2020-02-06:
Thank you for your reply. However, when I adjusted the weight ( object_velocity_BA_weight=1), the performance is still very poor.
[image]

GuaguaguaLiu, 2020-02-06:
In addition, I try to set the parameter 'whether_detect_object=false' (this parameter is mainly used in 'DetectCuboid' function in Tracking.cc ), thus the 'detect_3d_cuboid::detect_cuboid' is used to detect 3D object online.  However, when I test on the datatset '0018' your provided, the performance is really poor. Therefore, I compare the offline data 'pred_3d_obj_matched_tracked_txt' with the online results obtained by your 'detect_3d_cuboid::detect_cuboid' function, and I find they are quite different, especially 'yaw' angle. If the online results of 'detect_3d_cuboid::detect_cuboid' function is not as accurate as the Kitti offline data, resulting in poor performance?