shichaoy/cube_slam #29: Some error while catkin_make

ValeriiaIZH, 2020-06-02:
When i do catkin_make i have this errors. But when i add `include <numeric>` and `include <vector>` i have the same problem.

> In file included from /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:1:0:
> 
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:47:101: error: ‘std::vector’ has not been declared
>  bool read_obj_detection_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &strings);
>                                                                                                      ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:47:107: error: expected ‘,’ or ‘...’ before ‘<’ token
>  bool read_obj_detection_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &strings);
>                                                                                                            ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:49:102: error: ‘std::vector’ has not been declared
>  bool read_obj_detection2_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &strings);
>                                                                                                       ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:49:108: error: expected ‘,’ or ‘...’ before ‘<’ token
>  bool read_obj_detection2_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &strings);
>                                                                                                             ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:52:52: error: ‘std::vector’ has not been declared
>  void sort_indexes(const Eigen::VectorXd &vec, std::vector<int> &idx, int top_k);
>                                                     ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:52:58: error: expected ‘,’ or ‘...’ before ‘<’ token
>  void sort_indexes(const Eigen::VectorXd &vec, std::vector<int> &idx, int top_k);
>                                                           ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:53:52: error: ‘std::vector’ has not been declared
>  void sort_indexes(const Eigen::VectorXd &vec, std::vector<int> &idx);
>                                                     ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:53:58: error: expected ‘,’ or ‘...’ before ‘<’ token
>  void sort_indexes(const Eigen::VectorXd &vec, std::vector<int> &idx);
>                                                           ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:60:30: error: ‘vector’ in namespace ‘std’ does not name a template type
>  void print_vector(const std::vector<T> &vec);
>                               ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:60:36: error: expected ‘,’ or ‘...’ before ‘<’ token
>  void print_vector(const std::vector<T> &vec);
>                                     ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:64:51: error: ‘std::vector’ has not been declared
>  void linespace(T starting, T ending, T step, std::vector<T> &res);
>                                                    ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:64:57: error: expected ‘,’ or ‘...’ before ‘<’ token
>  void linespace(T starting, T ending, T step, std::vector<T> &res);
>                                                          ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:236:101: error: ‘std::vector’ has not been declared
>  bool read_obj_detection_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &all_strings)
>                                                                                                      ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:236:107: error: expected ‘,’ or ‘...’ before ‘<’ token
>  bool read_obj_detection_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &all_strings)
>                                                                                                            ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp: In function ‘bool read_obj_detection_txt(std::__cxx11::string, Eigen::MatrixXd&, int)’:
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:238:37: error: no matching function for call to ‘std::basic_ifstream<char>::basic_ifstream(const string&)’
>      if (!std::ifstream(txt_file_name))
>                                      ^
> In file included from /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:9:0:
> /usr/include/c++/5/fstream:495:7: note: candidate: std::basic_ifstream<_CharT, _Traits>::basic_ifstream(const char*, std::ios_base::openmode) [with _CharT = char; _Traits = std::char_traits<char>; std::ios_base::openmode = std::_Ios_Openmode]
>        basic_ifstream(const char* __s, ios_base::openmode __mode = ios_base::in)
>        ^
> /usr/include/c++/5/fstream:495:7: note:   no known conversion for argument 1 from ‘const string {aka const std::__cxx11::basic_string<char>}’ to ‘const char*’
> /usr/include/c++/5/fstream:481:7: note: candidate: std::basic_ifstream<_CharT, _Traits>::basic_ifstream() [with _CharT = char; _Traits = std::char_traits<char>]
>        basic_ifstream() : __istream_type(), _M_filebuf()
>        ^
> /usr/include/c++/5/fstream:481:7: note:   candidate expects 0 arguments, 1 provided
> /usr/include/c++/5/fstream:455:11: note: candidate: std::basic_ifstream<char>::basic_ifstream(const std::basic_ifstream<char>&)
>      class basic_ifstream : public basic_istream<_CharT, _Traits>
>            ^
> /usr/include/c++/5/fstream:455:11: note:   no known conversion for argument 1 from ‘const string {aka const std::__cxx11::basic_string<char>}’ to ‘const std::basic_ifstream<char>&’
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:243:5: error: ‘all_strings’ was not declared in this scope
>      all_strings.clear();
>      ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp: At global scope:
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:274:102: error: ‘std::vector’ has not been declared
>  bool read_obj_detection2_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &all_strings)
>                                                                                                       ^
[... 245 lines cut ...]
>                                    ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/detect_3d_cuboid.h:79:29: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
>        double max_cut_skew = 3;
>                              ^
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/box_proposal_detail.cpp: In member function ‘void detect_3d_cuboid::detect_cuboid(const cv::Mat&, const Matrix4d&, const MatrixXd&, Eigen::MatrixXd, std::vector<std::vector<cuboid*> >&)’:
> /home/valeriia/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/box_proposal_detail.cpp:531:55: error: ‘iota’ was not declared in this scope
>    iota(sort_idx_small.begin(), sort_idx_small.end(), 0);
>                                                        ^
> CMakeFiles/detect_3d_cuboid.dir/build.make:86: recipe for target 'CMakeFiles/detect_3d_cuboid.dir/src/object_3d_util.cpp.o' failed
> make[2]: *** [CMakeFiles/detect_3d_cuboid.dir/src/object_3d_util.cpp.o] Error 1
> CMakeFiles/detect_3d_cuboid.dir/build.make:62: recipe for target 'CMakeFiles/detect_3d_cuboid.dir/src/box_proposal_detail.cpp.o' failed
> make[2]: *** [CMakeFiles/detect_3d_cuboid.dir/src/box_proposal_detail.cpp.o] Error 1
[... cut ...]
> Makefile:138: recipe for target 'all' failed
> make: *** [all] Error 2
> <== Failed to process package 'detect_3d_cuboid': 
>   Command '['/home/valeriia/cubeslam_ws/devel_isolated/tictoc_profiler/env.sh', 'make', '-j6', '-l6']' returned non-zero exit status 2
> 
> Reproduce this error by running:
> ==> cd /home/valeriia/cubeslam_ws/build_isolated/detect_3d_cuboid && /home/valeriia/cubeslam_ws/devel_isolated/tictoc_profiler/env.sh make -j6 -l6
> 
> Command failed, exiting.

Ubuntu 16.04, ROS KInetic, OpenCV 3.4

cjzhang916, 2020-06-02:
I have the same error.
Ubuntu 18.04 ros melodic opencv 3.4.10

better629, 2020-06-22:
ubuntu 18.02 
ros melodic 
opencv 3.2.0

with `include <numeric>` is ok. but error with self-built opencv
```
/usr/local/lib/libopencv_imgproc.so.3.2.0：对‘powf@GLIBC_2.27’未定义的引用
/usr/local/lib/libopencv_core.so.3.2.0：对‘logf@GLIBC_2.27’未定义的引用
```

maybe due to the conda env

Royal-Wh1t3zZ, 2021-01-06:
I have the same error.
I think the version of g++ may cause these errors.

zhezhou1993, 2021-12-02:
My dependency fix here (check the most recent commit, but not for opencv error if you have any):
https://github.com/zhezhou1993/cube_slam/

GRF-Sunomikp31, 2022-07-21:
I wrote a tutorial on how to install cube-slam on Ubuntu 18.04, but in Chinese. If you have any questions, please contact me via the email on the tutorial.
https://blog.csdn.net/qq_44847636/article/details/125899174