shichaoy/cube_slam #30: object_slam run error

wangzhengboo, 2020-06-02:
when I run roslaunch object_slam object_slam_example.launch, the error I encountered is as follows.

[object_slam_node-1] process has died [pid 18032, exit code -11, cmd /home/wzb/cubeslam_ws/devel/lib/object_slam/object_slam_node __name:=object_slam_node __log:=/home/wzb/.ros/log/b7f52834-a492-11ea-8ac7-7cb27d1a2d0e/object_slam_node-1.log].
log file: /home/wzb/.ros/log/b7f52834-a492-11ea-8ac7-7cb27d1a2d0e/object_slam_node-1*.log

And I find the code exit in cube_slam/detect_3d_cuboid/src/object_3d_util.cpp line 304:

merge_lines_out = all_lines;

And when I use gdb to backtrace, The displayed content is as follows：

Thread 1 "object_slam_nod" received signal SIGSEGV, Segmentation fault.
0x00000000004444a0 in void Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::internal::assign_op<double> >(Eigen::Matrix<double, -1, -1, 0, -1, -1>&, Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, Eigen::internal::assign_op<double> const&) ()
(gdb) bt
#0  0x00000000004444a0 in void Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::internal::assign_op<double> >(Eigen::Matrix<double, -1, -1, 0, -1, -1>&, Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, Eigen::internal::assign_op<double> const&) ()
#1  0x00007ffff45cd21c in merge_break_lines(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, Eigen::Matrix<double, -1, -1, 0, -1, -1>&, double, double, double)
    () from /home/wzb/cubeslam_ws/devel/lib/libdetect_3d_cuboid.so
#2  0x00007ffff45c13b1 in detect_3d_cuboid::detect_cuboid(cv::Mat const&, Eigen::Matrix<double, 4, 4, 0, 4, 4> const&, Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, Eigen::Matrix<double, -1, -1, 0, -1, -1>, std::vector<std::vector<cuboid*, std::allocator<cuboid*> >, std::allocator<std::vector<cuboid*, std::allocator<cuboid*> > > >&) ()
   from /home/wzb/cubeslam_ws/devel/lib/libdetect_3d_cuboid.so
#3  0x000000000042f7f4 in incremental_build_graph(Eigen::Matrix<double, -1, -1, 0, -1, -1>&, Eigen::Matrix<double, -1, -1, 0, -1, -1>&, Eigen::Matrix<double, -1, -1, 0, -1, -1>&) ()
#4  0x0000000000421712 in main ()

Could you please help me to solve the problem? Thank you very much!

wangzhengboo, 2020-06-02:
I have solved the problem following the step in https://github.com/colmap/colmap/issues/67. However, I can not run the node as well.
I find the code exit at cube_slam/object_slam/src/main_obj.cpp line 562:
graph.optimize(5); // do optimization!

The error I encountered is as follows:
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[object_slam_node-1] process has died [pid 22119, exit code -6, cmd /home/wzb/cubeslam_ws/devel/lib/object_slam/object_slam_node __name:=object_slam_node __log:=/home/wzb/.ros/log/b7f52834-a492-11ea-8ac7-7cb27d1a2d0e/object_slam_node-1.log].
log file: /home/wzb/.ros/log/b7f52834-a492-11ea-8ac7-7cb27d1a2d0e/object_slam_node-1*.log

And when I use gdb to backtrace, the displayed content is as follows:
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Thread 1 "object_slam_nod" received signal SIGABRT, Aborted.
0x00007ffff0f48428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff0f48428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff0f4a02a in __GI_abort () at abort.c:89
#2  0x00007ffff158284d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff15806b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff1580701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff1580919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x0000000000434a42 in Eigen::internal::throw_std_bad_alloc() ()
#7  0x0000000000434abe in Eigen::internal::aligned_malloc(unsigned long) ()
#8  0x000000000044d3c5 in void std::vector<g2o::SE3Quat, Eigen::aligned_allocator<g2o::SE3Quat> >::_M_emplace_back_aux<g2o::SE3Quat const&>(g2o::SE3Quat const&) ()
#9  0x00007ffff33407b8 in g2o::SparseOptimizer::push(std::vector<g2o::OptimizableGraph::Vertex*, std::allocator<g2o::OptimizableGraph::Vertex*> >&) () from /home/wzb/cubeslam_ws/src/cube_slam/object_slam/Thirdparty/g2o/lib/libg2o.so
#10 0x00007ffff334a6cf in g2o::OptimizationAlgorithmLevenberg::solve(int, bool) ()
   from /home/wzb/cubeslam_ws/src/cube_slam/object_slam/Thirdparty/g2o/lib/libg2o.so
#11 0x00007ffff3341e27 in g2o::SparseOptimizer::optimize(int, bool) ()
   from /home/wzb/cubeslam_ws/src/cube_slam/object_slam/Thirdparty/g2o/lib/libg2o.so
#12 0x000000000042c3ec in incremental_build_graph(Eigen::Matrix<double, -1, -1, 0, -1, -1>&, Eigen::Matrix<double, -1, -1, 0, -1, -1>&, Eigen::Matrix<double, -1, -1, 0, -1, -1>&) ()
#13 0x00000000004211ed in main ()

TINY-KE, 2021-03-01:
请问 这个问题你解决了吗？

Augusthyq, 2022-04-08:
请问问题解决了吗

lin-fangzhou, 2023-04-03:
Excuse me, have you solved this problem? I have encountered the same problem