shichaoy/cube_slam #31: Does 2d detection need depth？

ns15417, 2020-09-25:
HI，Thanks for your project. Recently, i am trying to run it on my own dataset  and there are some part i am not quiet understand. PS. i already successfully ran all demo on my computer.
Take object_slam_example.launch for example. i make my own data as the code said: raw_imgs for original RGB images, filter_2d_obj_txts for 2D detection of images, truth_cam_poses.txt for camera poses(in which only the first pose is right); when running object_slam_example.launch, i set online_detect_mode to true cuz i want the online cube detection. 
And as the note said on main_obj.cpp, detect_cuboids_saved.txt and pop_cam_poses_saved.txt are not used in online detection, so i didn't read these two files.
My code runs well but stuck at publish_all_poses() and it reads the depth image. so does this mean that it needs RGBD dataset for tansferring 2D detection to 3D cube? or the depth image is only used for visualization?
i am not quiet understand what kinds of data do i need to prepare for online and offline. and i didn't find any files to clarify this. so i am still confused.
Thanks for your time and I really appreciate your open-source code. Thanks!!

UranusSong, 2020-09-25:
据我所知，CubeSLAM是不需要深度图像信息的，你的猜测是对的，深度图是用于demo的可视化。
另外，你跑的demo是object_slam模块吧，实际上，你需要跑的是orb_object_slam，那个完全不需要深度图像。

ns15417, 2020-09-25:
@UranusSong 
谢谢你的解答！！我也是才发现我可能跑错example了，重新跑一下试试
Thanks for your answer. I just noticed that i might run the wrong example. 
Thank you so much !!

closed 2020-09-25 by ns15417

deeprun, 2021-07-22:
I was able to run KITTI seq 07 dataset with curb-orb-slam. I even change the 3D data file to 2D data file, it also ran fine. However, when I used my own dataset, the result was not good. The 2D detection box was not even shown on the images. How was your results with your own data?