shichaoy/cube_slam #32: What Does InitToGround mean??

ns15417, 2020-09-29:
Hi, I am so happy i run orb_object_slam successfully and Thanks for your work;
While Running on my own dataset, i found the whole size turned smaller and smaller, especially whrn some object are detected. SO i was wondering are there any other parameters which may affect the scale of the whole map?
I found InitToGround in Tracking.cpp and i am not sure what this parameter means? Transfering points in camera to Ground? I am not sure what the Ground Coordinate looks like? Front is x, Left is Y,and up is Z (Like Odom)? What is the camera Coordinate looks like? is it the same as OrbSlam, it is Front is z, Right is x, and Down is y. ?
I am confused because of the InitToGround of kitti in Tracking::Tracking() in which InitToGround Quaternion is set to (0.7071,0,0,-0.7071) which in Eular Angle is (z,y,x) = (0,0,90). Why the relation between this two coordinates is rotate 90 around x-axis ?

shichaoy, 2020-10-05:
hi @ns15417   initToGround is the initial camera pose in ground world frame, ie T_world_camera, same as your understanding, it transforms a camera point to world frame.   camera defined as x right, y down, z forward.   ground is x right, y forward, z upward.
so for kitti style data where camera is parallel to ground,  the initToGround is 
[1   0   0  0
 0   0   1  0
 0  -1   0  height
 0   0   0  1]

can refer to coordinate system in second page of this  https://github.com/shichaoy/matlab_cuboid_detect/blob/master/illustrations.pdf

https://github.com/shichaoy/matlab_cuboid_detect/blob/master/illustrations.pdf

closed 2020-10-10 by ns15417

mmahdavian, 2020-10-28:
@ns15417  Hi. Can I know how long did it take for you to change the codes and use them according to your dataset?!
Thank You

ns15417, 2020-11-03:
i think it one or two week. From the understanding of the code to the processing of my own dataset