shichaoy/cube_slam #33: KeyFrame stuck together

ns15417, 2020-10-10:
Hi, @shichaoy thanks for your work.
The size of the objects turned to be normal after the issue: #32 
But I am still not sure why sometimes the size of the map turns to be smaller, I am wonder is this the problem of all monoSLAM or some parameter setting of cubeSLAM?

I gave an example result on my own dataset, and the order of these images are from 1 to 4. As the images showed, the size get smaller and smaller while sometimes it can be corrected by LOOPCLOSing. But without LoopClosing(as shown in image 4), the keyFrames just stuck at a same place. Are there any suggestion for this situation?

Please check the frames at : https://github.com/ns15417/cube_slam/tree/func/DebugMySequence/Result

Thanks for your work and time :)

UranusSong, 2020-10-13:
hey 兄弟，请问你的这个数据集是自己采集室内数据集吗？单目视觉slam有两个特点，一是尺度不确定性，即无法确定绝对尺度，二是误差累积造成的尺度漂移。
第一个问题可以通过给定先验值的办法解决，比如cubeslam中提到利用相机高度，并且检测出的object position/dimension都依赖于相机高度提供的尺度信息（前提是检测的object必须位于地面上，即**ground object**，不知道你的数据集是否满足这个条件）
第二个问题解决起来就比较困难了，因为即使第一帧提供了真实相机高度，仍然无法避免后续的误差累计造成尺度漂移，比如你的地图size越来越小，传统方法几乎只有回环矫正这一种。据我所知，cubeslam有两种抵抗尺度漂移的方案，并且都是针对KITTI数据集（其它自动驾驶数据集理论上也可以）。
+ 第一种是给object dimension设定先验值，每次检测到新的车辆对象后用先验值更新它的尺寸，最终这个尺寸可以反映在local optimization中，维持整个地图的尺度。
+ 第二个方案是为了弥补车辆对象过少的情况，利用相机高度值不断矫正尺度，详见 (Tracking.cc/CreateNewKeyframe)。
+ 总结：cubeslam在室外的优化效果更好，室内一般不具有ground object和锚定高度相机的配置，因此效果可能不是那么好。

***

Hey bro,  your own dataset is indoor right? Mono visual slam has two issues: one is scale uncertainty, another is sclae drift caused by error accumulation.  The former can be addressed by offering absolute scale metric clues like camera height or object dimension. However, the latter is hard to overcome. It can not avoid scale drift though you have given a right scale factor at the first frame, unless you keep providing absolute right value.  Otherwise, you have to depend on loop closing. 
As to my learning, CubeSLAM has two schemes for resisting scale drift:
+ The first one is to update detected 3D objects' size with prior length, width and height while run tracking. The prior size value will back propogate into local optimization ，so that keep map scale consistency.
+ The sceond one is deal with the scenario with too few vehicles, It will correct map scale with camera height ( so this method is only  for robot with anchored camera height . Details refer to Tracking.cc/CreateNewKeyFrame function)
+ Conclusion: Cubeslam performs better in outdoor. Indoor scenario generally does not contain much ground objects and the camera is not anchored, so the performence may not be so good.

ns15417, 2020-10-14:
@UranusSong 谢谢你的回答,这个数据集是我自己采集的室内数据集,物品也都是地面物体,比如瓶子,小箱子,毛巾等, 但是都不是很大的物体,相机的话是固定安装在机器人上的,有一个准确的初始位姿; 
我之前跑纯粹的orbslam的时候确实也遇到过在旋转的时候关键帧堆积的现象,不知道原因和cubeslam size变小的原因是否一样,所以比较困惑. 
根据你的回答,我想了解一下,cubeslam是在大的物体的时候效果更好吗?不知道有没有相关的测试或结论,也为你说outdoors,我可以这样理解吗? 另外,是否有推荐的优化方向呢?比如加入odom的信息来约束尺度,不知效果会如何

UranusSong, 2020-10-14:
+ 我猜的关键帧堆积与地图尺寸变小有可能是一个原因：相机旋转的时候，图像重叠度小，插入关键帧比较多是正常的（https://www.cnblogs.com/yepeichu/p/10784265.html）。并且在相机旋转大，平移小的情况下，单目slam容易发生运动退化，三角化所需的视差小，精度差，整体优化的时候更容易发生尺度漂移，也就是你所看到的map size变小了。
也许你在控制机器人的时候转弯平滑一些，转弯角小一些可以缓解这个问题。
+ cuebslam在理论上并不是大物体的时候表现更好，如果你可以在小尺度数据集上也充分利用到object size和相机高度先验信息，也可以达到相当的效果。只是cubeslam在KITTI上完成了这个工作（毕竟KITTI中主要的object就是车辆，object size比较好统计）。
+ 据我所知，orbslam+odom的工作是已经有的了，加入IMU或者直接用双目相机都算有意义的工作，不过这就和单目slam不是同一赛道了。

ns15417, 2020-10-14:
谢谢你的回答,清楚了很多,我目前已经完成了odom+普通SLAM的工作,想着可不可以进一步做odom+cubeslam,类似于双重计算scale的情况;

shichaoy, 2020-10-19:
多谢 @UranusSong 的回答，非常正确    It's hard to get correct size of mono slam, ususally need to have some assumption.  usually object/ceiling size, or camera height to ground.   for indoor dataset, I sometimes also meet the smaller scale problem. If it does happen for example in your image, cannot recover from it except loop closure correct it.   I usually try to get a good initial heigh estimation, when collecting data, don't have too aggressive rotations/translation, then visual slam itself usually doesn't have large drift, and I typically set small object size variance once detected, basically hope object can reduce the drift.  but this also requires the object detection is accurate.    Plane can also help reduce the drift.

ns15417, 2020-10-19:
@shichaoy by Plane, do you mean the Ground? i can get the segmentation of the Ground and it is kind of accurate, but i am not sure how to add this to the Optimization or the size ? I mean how to use the Ground or Combine Segmentation with Visual SLAM

shichaoy, 2020-10-21:
@ns15417 just based on ground plane in theory cannot resolve the scale issue. it may help a bit, there are different research on that, not straighforward to directly use
1. if you roughly have ground height estimation, you can unproject to get 3D depth, as initialization for points. more learning research on it.   https://shichaoy.github.io/Publications/iros_2016_popslam.pdf
2. the walls can also provide some geometric constraints in many views, while points/objects are only observed in a few frames.
https://shichaoy.github.io/Publications/2019_object_plane.pdf