shichaoy/cube_slam #34: error while launching orb_object_slam

mmahdavian, 2020-10-26:
Hi

I am trying to use "roslaunch orb_object_slam mono.launch" command to use orb_object_slam. I have changed the address in mono.launch and I have attached it here. When I use the command I get following error:
```
mohammad@mohammad:~/cubeslamws$ roslaunch orb_object_slam mono.launch 
... logging to /home/mohammad/.ros/log/fd8e5456-17bf-11eb-8dec-5c80b69e8dce/roslaunch-mohammad-31764.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mohammad:45929/

SUMMARY
========

PARAMETERS
 * /associate_point_with_object: True
 * /ba_dyna_obj_cam: True
 * /ba_dyna_obj_velo: True
 * /ba_dyna_pt_obj_cam: True
 * /base_data_folder: /home/shichao/ysc...
 * /build_worldframe_on_ground: True
 * /bundle_object_opti: True
 * /camera_object_BA_weight: 2.0
 * /draw_map_truth_paths: True
 * /draw_nonlocal_mappoint: True
 * /enable_ground_height_scale: True
 * /enable_loop_closing: False
 * /enable_viewer: True
 * /enable_viewimage: True
 * /enable_viewmap: True
 * /ground_dist_ratio: 0.08
 * /ground_everyKFs: 10
 * /ground_inlier_pts: 20
 * /ground_roi_lower: 3.0
 * /ground_roi_middle: 3.0
 * /init_qw: 0.7071
 * /init_qx: -0.7071
 * /init_qy: 0
 * /init_qz: 0
 * /init_x: 0
 * /init_y: 0
 * /init_z: 1.7
 * /mono_allframe_Obj_depth_init: False
 * /mono_firstframe_Obj_depth_init: False
 * /mono_firstframe_truth_depth_init: False
 * /obj_det_2d_thre: 0.5
 * /object_velocity_BA_weight: 0.5
 * /parallel_mapping: True
 * /remove_dynamic_features: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.16
 * /scene_name: kitti
 * /triangulate_dynamic_pts: False
 * /use_dynamic_klt_features: False
 * /use_truth_trackid: False
 * /whether_detect_object: True
 * /whether_dynamic_object: False
 * /whether_read_offline_cuboidtxt: True
 * /whether_save_final_optimized_cuboids: False
 * /whether_save_online_detected_cuboids: False

NODES
  /mono/
    image_proc (image_proc/image_proc)
  /
    ros_mono (orb_object_slam/ros_mono)

auto-starting new master
process[master]: started with pid [31776]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fd8e5456-17bf-11eb-8dec-5c80b69e8dce
process[rosout-1]: started with pid [31789]
started core service [/rosout]
process[mono/image_proc-2]: started with pid [31806]
process[ros_mono-3]: started with pid [31807]
Base_data_folder:  /home/shichao/ysc_space/dataset/processed_third/slam/kitti/kitti_odom/seq_07
[ WARN] [1603739893.320635552]: Turn off global loop closing!!
Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
[ros_mono-3] process has died [pid 31807, exit code -11, cmd /home/mohammad/catkin_ws/devel/lib/orb_object_slam/ros_mono /home/mohammad/catkin_ws/src/cube_slam/orb_object_slam/Vocabulary/ORBvoc.bin /home/mohammad/catkin_ws/src/cube_slam/orb_object_slam/Examples/Monocular/KITTI04-12_me.yaml /camera/image_raw:=/kitti/left/image_raw __name:=ros_mono __log:=/home/mohammad/.ros/log/fd8e5456-17bf-11eb-8dec-5c80b69e8dce/ros_mono-3.log].
log file: /home/mohammad/.ros/log/fd8e5456-17bf-11eb-8dec-5c80b69e8dce/ros_mono-3*.log

```
I don't know why am I getting this error. Also the Base_data_folder is still wrong. Although I have changed it. Also, when I go to the cube_slam/orb_object_slam/launch folder and run the mono.launch file, the address gets corrected, but I still get the same errors. Do you know what might be the problem here?

[mono.launch.zip](https://github.com/shichaoy/cube_slam/files/5440933/mono.launch.zip)

mmahdavian, 2020-10-26:
Also in mono.launch file there is "args="$(find orb_object_slam)/Vocabulary/ORBvoc.bin". But there is no ORBvoc.bin file in the Vocabulary folder.

UranusSong, 2020-10-27:
@mmahdavian Hi！I have  met the same error before. As you said,  there is no ORBvoc.bin file in the Vocabulary folder, which caused this problem. I addressed this issue by downloading ORBvoc.txt from official orb-slam2 repo and fixing the corresponding path argument.

mmahdavian, 2020-10-27:
@UranusSong Thanks for your comment. I made that change, but I am still getting the same error. I don't get why am I still seeing this : Base_data_folder:  /home/shichao/ysc_space/dataset/processed_third/slam/kitti/kitti_odom/seq_07
I have changed the address in the launch file.

mmahdavian, 2020-10-27:
The problem is fixed. I just needed to use "source devel/setup.bash" again.
Thank You

closed 2020-10-27 by mmahdavian