shichaoy/cube_slam #36: pop_pose_to_ground is not initialized when using online cuboid proposal method

xvzezi, 2020-11-06:
While reading through the code in Tracking.cc
In function ```void Tracking::DetectCuboid(KeyFrame *pKF)```
In the code block where using online cuboid proposal method, between line 1615 to 1621
```c++
cv::Mat frame_pose_to_init = pKF->GetPoseInverse();
cv::Mat frame_pose_to_ground = frame_pose_to_init;
if(!build_worldframe_on_ground) frame_pose_to_ground = InitToGround * frame_pose_to_ground;
Eigen::Matrix4f cam_transToGround = Converter::toMatrix4f(pop_pose_to_ground);    // ??? pop_pose_to_ground is not given and value, I assume that the frame_pose_to_ground is assigned to pop_pose_to_ground, am I right?
detect_cuboid_obj->detect_cuboid(pKF->raw_img,cam_transToGround.cast<double>(), ....)
```
SO, is the pop_pose_to_ground intentionally left empty here?
Thanks for all your help.

shichaoy, 2020-11-09:
hi @xvzezi I think you are correct, this is a big mistake, I usually just use "whether_read_offline_cuboidtxt" case, didn't notice this online detection problem. I think it should be
```
Eigen::Matrix4f cam_transToGround = Converter::toMatrix4f(frame_pose_to_ground);
```
as there are a few lines in front of it to compute "frame_pose_to_ground".      I haven't run it for long time, when you print cam_transToGround, is it fixed default value?

xvzezi, 2020-11-09:
@shichaoy  I am still testing on the codes. The function ```Converter::toMatrix4f()``` does not has an empty matrix check inside, and directly access the values. So in the old way, there should be access errors. I will let you know when tested.

shichaoy, 2020-11-09:
@xvzezi  thanks! I will also test and look recently. I think should assign  pop_pose_to_ground= frame_pose_to_ground,  as frame_pose_to_ground is also used later.

xvzezi, 2020-11-09:
@shichaoy yes, one should assign it
```
pop_pose_to_ground = frame_pose_to_ground
```
because ```pop_pose_to_ground``` is used later.

And a little piece of advice, initToGround is hard to understand. I prefer Tgc (T_ground_camera) because it fits the ORB_SLAM2's rules and may make the codes much easier to catch up with.

xvzezi, 2020-11-09:
Sorry, it seems that i shouldn't close this before the bug is fixed.