shichaoy/cube_slam #38: Matlab YOLO2  sample output

Lauenburg, 2021-01-05:
**What?**
Can you please provide a sample file or even a meta description of a 2D object bounding box txt files?

**Why?**
I would like to run the mono example online using my own 2D object bounding box txt files. 
But as I do not have Matlab I can not figure out what format a file `/mats/filter_match_2d_boxes_txts/<frame_index_c>__yolo2_0.2.txt` should have.

Lauenburg, 2021-01-05:
Ok, I think I figured out the data format:
class_name(1), bounding_box(2-5), confidence_score(6)
- The confidence score is not necessary as it is currently hardcoded to one. 
- The bounding box is represented as follows: left_x(1), top_y(2), obj_width(3), obj_height(4)
Is that correct?

deeprun, 2021-07-20:
Did you solve this issue?