shichaoy/cube_slam #39: catkin_make error.

yanghan1ros, 2021-01-10:
[ 40%] Building CXX object cube_slam/line_lbd/CMakeFiles/linelbd_detect_node.dir/src/detect_lines.cpp.o
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/object_3d_util.cpp: In function ‘void fuse_normalize_scores_v2(const VectorXd&, const VectorXd&, Eigen::VectorXd&, std::vector<int>&, double, bool)’:
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/object_3d_util.cpp:503:14: error: ‘iota’ is not a member of ‘std’
         std::iota(dist_sorted_inds.begin(), dist_sorted_inds.end(), 0);
              ^~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/object_3d_util.cpp:503:14: note: suggested alternative: ‘not2’
         std::iota(dist_sorted_inds.begin(), dist_sorted_inds.end(), 0);
              ^~~~
              not2
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/object_3d_util.cpp:529:14: error: ‘iota’ is not a member of ‘std’
         std::iota(final_keep_inds.begin(), final_keep_inds.end(), 0);
              ^~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/object_3d_util.cpp:529:14: note: suggested alternative: ‘not2’
         std::iota(final_keep_inds.begin(), final_keep_inds.end(), 0);
              ^~~~
              not2
In file included from /home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:1:0:
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:47:101: error: ‘std::vector’ has not been declared
 bool read_obj_detection_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &strings);
                                                                                                     ^~~~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:47:107: error: expected ‘,’ or ‘...’ before ‘<’ token
 bool read_obj_detection_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &strings);
                                                                                                           ^
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:49:102: error: ‘std::vector’ has not been declared
 bool read_obj_detection2_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &strings);
                                                                                                      ^~~~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:49:108: error: expected ‘,’ or ‘...’ before ‘<’ token
 bool read_obj_detection2_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &strings);
                                                                                                            ^
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:52:52: error: ‘std::vector’ has not been declared
 void sort_indexes(const Eigen::VectorXd &vec, std::vector<int> &idx, int top_k);
                                                    ^~~~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:52:58: error: expected ‘,’ or ‘...’ before ‘<’ token
 void sort_indexes(const Eigen::VectorXd &vec, std::vector<int> &idx, int top_k);
                                                          ^
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:53:52: error: ‘std::vector’ has not been declared
 void sort_indexes(const Eigen::VectorXd &vec, std::vector<int> &idx);
                                                    ^~~~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:53:58: error: expected ‘,’ or ‘...’ before ‘<’ token
 void sort_indexes(const Eigen::VectorXd &vec, std::vector<int> &idx);
                                                          ^
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:60:30: error: ‘vector’ in namespace ‘std’ does not name a template type
 void print_vector(const std::vector<T> &vec);
                              ^~~~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:60:36: error: expected ‘,’ or ‘...’ before ‘<’ token
 void print_vector(const std::vector<T> &vec);
                                    ^
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:64:51: error: ‘std::vector’ has not been declared
 void linespace(T starting, T ending, T step, std::vector<T> &res);
                                                   ^~~~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/include/detect_3d_cuboid/matrix_utils.h:64:57: error: expected ‘,’ or ‘...’ before ‘<’ token
 void linespace(T starting, T ending, T step, std::vector<T> &res);
                                                         ^
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:236:101: error: ‘std::vector’ has not been declared
 bool read_obj_detection_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &all_strings)
                                                                                                     ^~~~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:236:107: error: expected ‘,’ or ‘...’ before ‘<’ token
 bool read_obj_detection_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &all_strings)
                                                                                                           ^
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp: In function ‘bool read_obj_detection_txt(std::__cxx11::string, Eigen::MatrixXd&, int)’:
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:243:5: error: ‘all_strings’ was not declared in this scope
     all_strings.clear();
     ^~~~~~~~~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp: At global scope:
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:274:102: error: ‘std::vector’ has not been declared
 bool read_obj_detection2_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &all_strings)
                                                                                                      ^~~~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:274:108: error: expected ‘,’ or ‘...’ before ‘<’ token
 bool read_obj_detection2_txt(const std::string txt_file_name, Eigen::MatrixXd &read_number_mat, std::vector<std::string> &all_strings)
                                                                                                            ^
[... 126 lines cut ...]
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/matrix_utils.cpp:365:60: error: expected ‘,’ or ‘...’ before ‘<’ token
 template void linespace(double, double, double, std::vector<double> &);
                                                            ^
cube_slam/detect_3d_cuboid/CMakeFiles/detect_3d_cuboid.dir/build.make:88: recipe for target 'cube_slam/detect_3d_cuboid/CMakeFiles/detect_3d_cuboid.dir/src/matrix_utils.cpp.o' failed
make[2]: *** [cube_slam/detect_3d_cuboid/CMakeFiles/detect_3d_cuboid.dir/src/matrix_utils.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/box_proposal_detail.cpp: In member function ‘void detect_3d_cuboid::detect_cuboid(const cv::Mat&, const Matrix4d&, const MatrixXd&, Eigen::MatrixXd, std::vector<std::vector<cuboid*> >&)’:
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/box_proposal_detail.cpp:531:3: error: ‘iota’ was not declared in this scope
   iota(sort_idx_small.begin(), sort_idx_small.end(), 0);
   ^~~~
/home/a-y-x/cubeslam_ws/src/cube_slam/detect_3d_cuboid/src/box_proposal_detail.cpp:531:3: note: suggested alternative: ‘int’
   iota(sort_idx_small.begin(), sort_idx_small.end(), 0);
   ^~~~
   int
[ 42%] Linking CXX executable /home/a-y-x/cubeslam_ws/devel/lib/line_lbd/linelbd_detect_node
[ 42%] Built target linelbd_detect_node
[... cut ...]
cube_slam/detect_3d_cuboid/CMakeFiles/detect_3d_cuboid.dir/build.make:75: recipe for target 'cube_slam/detect_3d_cuboid/CMakeFiles/detect_3d_cuboid.dir/src/object_3d_util.cpp.o' failed
make[2]: *** [cube_slam/detect_3d_cuboid/CMakeFiles/detect_3d_cuboid.dir/src/object_3d_util.cpp.o] Error 1
CMakeFiles/Makefile2:1531: recipe for target 'cube_slam/detect_3d_cuboid/CMakeFiles/detect_3d_cuboid.dir/all' failed
make[1]: *** [cube_slam/detect_3d_cuboid/CMakeFiles/detect_3d_cuboid.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

yanghan1ros, 2021-01-10:
ubuntu18.04

315haoge, 2021-03-27:
Meet the same problem, but don't know how to sovle

zhezhou1993, 2021-12-02:
My dependency fix here (check the most recent commit):
https://github.com/zhezhou1993/cube_slam/

GRF-Sunomikp31, 2022-07-21:
I wrote a tutorial on how to install cube-slam on Ubuntu 18.04, but in Chinese. If you have any questions, please contact me via the email on the tutorial.
https://blog.csdn.net/qq_44847636/article/details/125899174