shichaoy/cube_slam #49: something wrong when launch the object_slam_example

notasadsong, 2022-04-20:
`... logging to /home/zwh/.ros/log/0c768bf8-c059-11ec-a2b4-d46a6a8b2053/roslaunch-zwh-Inspiron-7472-25422.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zwh-Inspiron-7472:36499/

SUMMARY
========

PARAMETERS
 * /base_folder: /home/zwh/catkin_...
 * /online_detect_mode: True
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /save_results_to_txt: False

NODES
  /
    object_slam_node (object_slam/object_slam_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [25432]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0c768bf8-c059-11ec-a2b4-d46a6a8b2053
process[rosout-1]: started with pid [25443]
started core service [/rosout]
process[object_slam_node-2]: started with pid [25446]
process[rviz-3]: started with pid [25447]

base_folder   /home/zwh/catkin_ws/src/cube_slam/object_slam/data/
[ WARN] [1650424933.163888663]: Online detect object mode !!

double free or corruption (out)
[object_slam_node-2] process has died [pid 25446, exit code -6, cmd /home/zwh/catkin_ws/devel/lib/object_slam/object_slam_node __name:=object_slam_node __log:=/home/zwh/.ros/log/0c768bf8-c059-11ec-a2b4-d46a6a8b2053/object_slam_node-2.log].
log file: /home/zwh/.ros/log/0c768bf8-c059-11ec-a2b4-d46a6a8b2053/object_slam_node-2*.log
`

Could you please tell me how to solve it?

lanhupoyx, 2022-07-07:
I got the same issue with you，have no idea.

lin-fangzhou, 2023-04-03:
Excuse me, have you solved this problem? I have encountered the same problem