shichaoy/cube_slam #50: Program running error

wangni-bupt, 2022-06-13:
An error is reported when "Found one bad object !!!!!!!!!!!!!!!!!!!!!!!!! " occurs
```
Detect cuboid for pKF id: 252  total id: 826  numObj: 7
begin to associate cuboids #candidate:   0   #landmarks   3   #localKFs   12
Found one bad object !!!!!!!!!!!!!!!!!!!!!!!!!  47  2  0
[ros_mono-3] process has died [pid 9501, exit code -11, cmd /home/wang/Total/cubeslam_ws/devel/lib/orb_object_slam/ros_mono /home/wang/Total/cubeslam_ws/src/cube_slam/orb_object_slam/Vocabulary/ORBvoc.bin /home/wang/Total/cubeslam_ws/src/cube_slam/orb_object_slam/Examples/Monocular/KITTI04-12_me.yaml /camera/image_raw:=/kitti/left/image_raw __name:=ros_mono __log:=/home/wang/.ros/log/41750dc4-ead5-11ec-a0eb-25bfe08d23ac/ros_mono-3.log].
log file: /home/wang/.ros/log/41750dc4-ead5-11ec-a0eb-25bfe08d23ac/ros_mono-3*.log
[ WARN] [1655096339.650441103]: The input topic '/mono/image_raw' is not yet advertised
[ WARN] [1655096339.650508607]: The input topic '/mono/camera_info' is not yet `advertised
```

NEUer-GaoWei, 2022-10-19:
I now have the same problem when running the code. Have you solved it? Looking forward to your reply~

NEUer-GaoWei, 2022-10-19:
> An error is reported when "Found one bad object !!!!!!!!!!!!!!!!!!!!!!!!! " occurs
>
> ```
> Detect cuboid for pKF id: 252  total id: 826  numObj: 7
> begin to associate cuboids #candidate:   0 [...]

I now have the same problem when running the code. Have you solved it? Looking forward to your reply

NEUer-GaoWei, 2022-10-21:
> An error is reported when "Found one bad object !!!!!!!!!!!!!!!!!!!!!!!!! " occurs
>
> ```
> Detect cuboid for pKF id: 252  total id: 826  numObj: 7
> begin to associate cuboids #candidate:   0 [...]

I have solved this problem，I changed $(find orb_object_slam)/Examples/Monocular/KITTI04-12_me.yaml"> in mono.launch to $(find orb_object_slam)/Examples/Monocular/KITTI04-12.yaml">

homocysteine, 2022-11-17:
@wangni-bupt Hi, Have you solve the problem? I met the same one. The system crashed when it found a bad object.

homocysteine, 2022-12-09:
I modified https://github.com/shichaoy/cube_slam/blob/c371989bf1448802898bc7f17e99bba4a76167e6/orb_object_slam/src/MapObject.cc#L225 from `pKF->EraseMapObjectMatch(mit->second)` to `pKF->EraseMapObjectMatch(this);`. It seems to solve the 'bad object' problem.

wangni-bupt, 2022-12-12:
I'm debugging with gdb, and it looks like I'm traversing the previous null pointer

homocysteine, 2022-12-15:
@wangni-bupt 我也是BUPT的，研究SLAM方向，我们可以加微信交流一下（vx: woshichenyushi），科研路上不孤独。